rhummelmose/BluetoothKit #4: OS X to iOS

lfb-cd, 2015-10-14:
May I ask some questions?
I want my iPhone connect to my mac.which one do you think should be my Peripheral?
And I want to keep running in background,so my mac app could connect with my iPhone(From a distance to close),how can I deal with this? thank you.

rasmusth, 2015-10-14:
Of course you may.

The device that is supposed to be connected to is the peripheral.

From a distance to close, how do you mean?

lfb-cd, 2015-10-14:
If take away my iPhone,Because I want my driver could connect

rasmusth, 2015-10-19:
Provided that you've set the appropriate values in your info.plist file, the connection will continue running in the background. I suggest you read the following: https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html

Once you lose a connection to a peripheral, you can attempt connecting again with a high timeout. Once in range of the peripheral again it will succeed in connecting. For what you're building however, it sounds as if you're going to need restoration, a concept that is not support by BluetoothKit. Maybe you will need to use CoreBluetooth directly instead. I suggest you read the section "Adding Support for State Preservation and Restoration" under the article linked above.

rasmusth, 2015-10-19:
Please close this issue if you find my answer okay.

yoavschwartz, 2015-10-19:
Did you think about adding support for it? Also maybe adding support for aggressive reconnection (sort of pairing) . I would love to help.

closed 2015-10-20 by lfb-cd

rasmusth, 2015-10-20:
It is a possibility but I am not sure how passing identifiers to the framework, when the application is launched with them should work. If you have ideas you're welcome to share.

Aggressive reconnection isn't a concept I'm familiar with, but I expect it to be self-explanatory, in which case you can simple try to connect with a short timeout and then try again repeatedly? I do not think this should be built into the framework.

yoavschwartz, 2015-10-20:
It's not a concept your'e familiar with because I couldn't find better words for it. I didn't want to use "Bonding" or "Pairing" because it would be misleading. But as per apple's recommendation, you could mimic Continuity or the iWatch type of connection, by asking to reconnect every time you disconnect from a device. Connection requests don't time out on purpose, so you could reconnect even a week later. Apple will only scan at opportune times so that battery isn't a large concern.  If you combine that with registering with an identifier, you could get woken up even if the app was killed. That way you can mimic pairing.
I just thought it would be nice if all this boiler plate could be abstracted away in BluetoothKit, so you would maybe call "pairPeripheralWithOptions" and choose if you want to register for notifications on reconnect for example even after the app is killed

I hope I managed to explain myself :)

rasmusth, 2015-10-20:
Okay, you did manage to explain yourself very well. You are correct, this is how you mimic the automatic pairing with BLE on iOS.

Maybe it would actually make sense to have a configuration option on a BKCentralConfiguration object to enable automatic reconnection after disconnects and an optional restoration identifier. As the connect call on BKCentral is concluded by a closure, we'd have to introduce a new delegate method, along the lines of remotePeripheralDidReconnect, and an option on connect to never time out.

These are good ideas. If you'd really like to help, I suggest we create issues that are specific to the work that needs be done and add them to the 0.3.0 milestone. I've already added 2 that I think appropriate.

yoavschwartz, 2015-10-20:
Yeah, I am currently working on a project with a bluetooth lock, I noticed that many times my code turns out similar to yours, I would love BluetoothKit to reach a place where I can use it to replace CoreBluetooth in my project :) I became so frustrated from Apple's delegate pattern and quirks I have even made an attempt at porting CoreBluetooth to PromiseKit ;) I will spend the weekend familiarizing myself with all your code.

rasmusth, 2015-10-20:
Sounds cool. Are you on LinkedIn or similar?

yoavschwartz, 2015-10-20:
On a side note, I think maybe theres also a place for a feature that would abstract away the re descovering flow of retrievePeripheralsWithIdentifiers -> retrieveConnectedPeripheralsWithServices -> scan for x seconds.
In my code I called it "findPeripheralWithTimeout" (In my code I use the device name, but I assume other people may have other needs so maybe it should accept a predicate of the form CBPeripheral -> Boolean.) This would also require BluetoothKit to cache all the UUID's of the devices you connected too, but that should be very simple.

yoavschwartz, 2015-10-20:
https://dk.linkedin.com/in/schwartzyoav

rasmusth, 2015-10-20:
I was actually considering this. I am not sure if I like the idea of the framework persisting things to disk, but it may be okay.

yoavschwartz, 2015-10-20:
I agree, I would probably put it under the BKCentralConfiguration, maybe a "shouldCacheUUIDs" or something, but I would enable it by default. After all, UUID's are device specific so it's junk info for anyone else. I assume most people will connect to a device more than one time and they don't care how you find it. And I don't think you need to go through the entire Apple doc to figure out that if another app connected to the BLE keyboard you won't find it by scanning ;)

rasmusth, 2015-10-20:
You are correct. In that case I think we should make a dedicated folder for it in the documents folder and make it possible to change the folder in which the defaults are saved.