rhummelmose/BluetoothKit #5: Support for Beacon Advertising

felix-dumit, 2015-10-20:
I couldn't find support for starting a Peripheral that advertises with iBeacon data.
Maybe there could be a `BKPeripheralConfiguration` for advertising iBeacon.

Also, is it currently possible to start advertising and already set the data for certain characteristics? I'm considering using this in my app, but in my scenario the peripheral doesn't send the data to the centrals, but rather makes available certain data via characteristics that centrals can read at any time.

Thanks!

rasmusth, 2015-10-20:
Hi Felix.

Thanks for your questions and proposals.

In order to broadcast as an iBeacon, it'd have to be possible to override the advertisement dictionary, probably on the BKPeripheralConfiguration. This could probably make sense, but I'm not sure if BluetoothKit in that case would be overkill compared to CoreBluetooth alone. What do you think?

It isn't possible to model characteristics as proposed by the Bluetooth specification. Instead, BluetoothKit manages one characteristic that allows you to send arbitrary data. This allows you to send archived objects easily provided that their classes conform to NSCoding.

I plan on expanding BluetoothKit with a combined peer class, that acts as both a central and peripheral at the same time, allowing one to pass arbitrary data both ways. I hope you can imagine how this will allow for object messaging in a cool way.

All thoughts are welcome :)

rasmusth, 2016-05-12:
Closing this.

closed 2016-05-12 by rasmusth