rhummelmose/BluetoothKit #8: tvos support

matteocrippa, 2015-10-31:
tvOS is pretty new and as far as I figured out some feature are not available (yet), but do you plan to add support for it next releases even for cocoapods auto install?

rasmusth, 2015-10-31:
Let me look if I can add a target for that quickly.

matteocrippa, 2015-10-31:
Yep, the unique thing is that I'm pretty aware of some functions available in iOS are disabled/removed in tvOS - https://developer.apple.com/library/prerelease/tvos/releasenotes/General/tvOS90APIDiffs/

rasmusth, 2015-10-31:
@matteocrippa I do not have the new Apple TV yet. But this new target does build. It only supports taking the central role though, as that is all that is allowed by tvOS.

rasmusth, 2015-10-31:
It won't work with Cocoapods. I'll have to split it into subspecs, which probably makes sense anyhow, as you'll most likely often only want to be either a peripheral or a central on one device.

matteocrippa, 2015-10-31:
@rasmusth i'll give a try next week back work with the atv dev kit, 'll keep you posted!

rasmusth, 2015-11-03:
@matteocrippa !

matteocrippa, 2015-11-03:
@rasmusth i was out of office in those last few days, hopefully tomorrow i'll be back in the office

rjgdesign, 2015-11-25:
Any update? Awesome library, would love to use it with tvOS! :)

rasmusth, 2015-12-03:
@rjgdesign Please check out the development branch and test if it works with the tvOS target. I don't have an Apple TV.

rasmusth, 2016-05-12:
Fixed.

commit 141dfa81 (2015-10-31): Target for tvOS. Closes #8.
closed 2016-05-12 by rasmusth