rhummelmose/BluetoothKit #10: BKPeriphial Protocol

kswain1, 2015-12-15:
I am following the example of setting up a peripheral, and I am stuck when adding the BKPeriphial to my view controller class. How do I implement the BKPeriphialController Protocol?

rasmusth, 2015-12-17:
Hi kswain1.

There is nothing in the framework called BKPeripheralController Protocol. What exactly are you referring to?

kswain1, 2015-12-18:
I had to add the BKPeripheralDelegate functions below to use a peripheral object in my class  

 // MARK: BKPeripheralDelegate
    
    internal func peripheral(peripheral: BKPeripheral, remoteCentralDidConnect remoteCentral: BKRemoteCentral) {
        Logger.log("Remote central did connect: \(remoteCentral)")
        sendDataBarButtonItem.enabled = true
    }
    
    internal func peripheral(peripheral: BKPeripheral, remoteCentralDidDisconnect remoteCentral: BKRemoteCentral) {
        Logger.log("Remote central did disconnect: \(remoteCentral)")
        sendDataBarButtonItem.enabled = false
    }

closed 2015-12-18 by kswain1