rhummelmose/BluetoothKit #9: 2 way communication

DigitalD1982, 2015-12-15:
Hi there,

As far as I can tell there is no way to have 2 way communication with BluetoothKit today. Is this possible and on the roadmap?

Thanks, enjoying BK so far! Just ran into a road block though with the 2 way communication.

rasmusth, 2015-12-17:
Hi @DigitalD1982.

First of all thanks for using the framework.

You are correct in your observation there. See #7.

Due to the fact that there is no support for CBCentralManager on tvOS though, I think I'll revise the approach and implement sending data from the BKCentral object as well. I will have to ensure though, that the throughput is good enough first. You are very welcome to help if you have the time.

If not you can always be both a BKCentral and a BKPeripheral on each device.

rasmusth, 2016-05-12:
See #15

closed 2016-05-12 by rasmusth