rhummelmose/BluetoothKit #11: Hi there !

tastyapp, 2016-01-18:
I really appreciate that you share this amazing looking library. I downloaded the example project and ran it on a real device. But there is an issue with the bluetooth "Unsupported" or "Unavailable". The bluetooth on the device is active and the error says: 
Error from scanning: InternalError(Optional(BluetoothKit.BKCentralStateMachine.Error.Transitioning(BluetoothKit.BKCentralStateMachine.State.Unavailable(BluetoothKit.BKUnavailabilityCause.Unsupported), [BluetoothKit.BKCentralStateMachine.State.Available])))

If you can help me, I would be thankful 
tastyapp

rasmusth, 2016-01-20:
Hi there.

In that case your device probably doesn't support Bluetooth LE. Which device do you have?

tastyapp, 2016-01-20:
I'm using an Iphone 6 and an Ipad 2

rasmusth, 2016-01-20:
The iPad 2 doesn't support Bluetooth LE. iPhone 6 does, please ensure that Bluetooth is on.

cguotju, 2016-04-01:
Same problem here on iPhone6. Plus, when I'm using the UUID 6E400001-B5A3-F393-­E0A9-­E50E24DCCA9E , it gives me a nil.

rasmusth, 2016-04-01:
@cguotju Which error do you get?

nsleader, 2016-04-05:
Same problem on iPhone 6 iOS 9.3.1

    let central = BKCentral()
    central.delegate = self
    do {
            let serviceUUID = NSUUID(UUIDString: "6E6B5C64-FAF7-40AE-9C21-D4933AF45B23")!
            let characteristicUUID = NSUUID(UUIDString: "477A2967-1FAB-4DC5-920A-DEE5DE685A3D")!
            let configuration = BKConfiguration(dataServiceUUID: serviceUUID, dataServiceCharacteristicUUID: characteristicUUID)
            try central.startWithConfiguration(configuration)
    } catch let error {
            print(error)
    }
        
    central.scanContinuouslyWithChangeHandler({ changes, discoveries in        
            print(discoveries.map { $0.localName } )
            }, stateHandler: { newState in
                print("State: ", newState)
            }, errorHandler: { error in
                print("Error from scanning: \(error)")
    })`

Error:

`Error from scanning: InternalError(Optional(BluetoothKit.BKCentralStateMachine.Error.Transitioning(BluetoothKit.BKCentralStateMachine.State.Starting, [BluetoothKit.BKCentralStateMachine.State.Available])))`

0x5e, 2016-05-01:
same problem. iPhone6 iOS 9.3

```
        let central = BKCentral()
        central.delegate = self
        do {
            let serviceUUID = NSUUID(UUIDString: "6E6B5C64-FAF7-40AE-9C21-D4933AF45B23")!
            let characteristicUUID = NSUUID(UUIDString: "477A2967-1FAB-4DC5-920A-DEE5DE685A3D")!
            let configuration = BKConfiguration(dataServiceUUID: serviceUUID, dataServiceCharacteristicUUID: characteristicUUID)
            try central.startWithConfiguration(configuration)
            // You are now ready to discover and connect to peripherals.
            central.scanContinuouslyWithChangeHandler({ (changes, discoveries) in
                for discovery in discoveries {
                    print(discovery)
                }
                }, stateHandler: { (newState) in
                    print("State: ", newState)
                }, errorHandler: { (error) in
                    print("Error from scanning: \(error)")
            })
            
        } catch let error {
            print(error)
        }
```

```
Error from scanning: InternalError(Optional(BluetoothKit.BKCentralStateMachine.Error.Transitioning(BluetoothKit.BKCentralStateMachine.State.Starting, [BluetoothKit.BKCentralStateMachine.State.Available])))
```

0x5e, 2016-05-01:
@tastyapp @nsleader @cguotju @rasmusth 
Found the problem. Scan can't be started immediately after `central.startWithConfiguration(configuration)`

if I startWithConfiguration in viewDidload, and scanContinuouslyWithChangeHandler in buttonClickAction, the error disappeared :-)

rasmusth, 2016-05-02:
If this is true I should add a completion handler to the start method. Thanks for the information.

jessepollak, 2016-05-10:
I had this same issue and adding a brief delay also worked!

gusthiele, 2016-05-10:
@jessepollak
Could you please indicate where did you add the delay?
Thanks!

jessepollak, 2016-05-10:
It looked something like this:

```swift
let central = BKCentral()

do {
        let serviceUUID = NSUUID(UUIDString: "6E6B5C64-FAF7-40AE-9C21-D4933AF45B23")!
        let characteristicUUID = NSUUID(UUIDString: "477A2967-1FAB-4DC5-920A-DEE5DE685A3D")!
        let configuration = BKConfiguration(dataServiceUUID: serviceUUID, dataServiceCharacteristicUUID: characteristicUUID)
        try central.startWithConfiguration(configuration)

        let delayTime = dispatch_time(DISPATCH_TIME_NOW, Int64(1 * Double(NSEC_PER_SEC)))
        dispatch_after(delayTime, dispatch_get_main_queue()) {
             central.scanContinuouslyWithChangeHandler({ changes, discoveries in        
                 print(discoveries.map { $0.localName } )
              }, stateHandler: { newState in
                 print("State: ", newState)
              }, errorHandler: { error in
                 print("Error from scanning: \(error)")
              })
        }
} catch let error {
        print(error)
}
```

gusthiele, 2016-05-11:
@jessepollak 
Thank you! I will try it.

rasmusth, 2016-05-12:
This isn't what you're supposed to do guys. You need to add yourselves as availability observers before calling start, once the Bluetooth stack is available for use after calling start, you will be notified. Waiting like you propose there is a hack and it is never guaranteed to become available because it relies on the underlying Bluetooth stack.

rasmusth, 2016-05-12:
In addition, this question has evolved a bit. Originally tastyapp asked why it failed for him, it did so because he was using a device without support for Bluetooth LE. Most of you others are experiencing the same kind of issue because you don't use the availability observation functionality.

rasmusth, 2016-05-12:
Corrected readme, sorry guys. See commit https://github.com/rasmusth/BluetoothKit/commit/b5751f134ab08a711efbe4067d984af261b9db87

closed 2016-05-12 by rasmusth

ibrahimuslu, 2017-01-22:
Error from scanning: internalError(Optional(BluetoothKit.BKCentralStateMachine.BKError.transitioning(BluetoothKit.BKCentralStateMachine.State.scanning, [BluetoothKit.BKCentralStateMachine.State.available])))

same problem in 0.4.0 version. 

Could you please give a hand to this problem.

ibrahimuslu, 2017-01-22:
Ok i found the problem is still there because you left scan() in viewDidAppear

rhummelmose, 2017-01-23:
@ibrahimuslu Correct. The thing is, it isn't actually a problem, because the error doesn't cause anything to fail.

qingqinghebiancao, 2017-10-23:
Just as @rhummelmose said, we should start scan operation when BLE is available.
```    
class CentralController: UIViewController {
        private func scan() {
          central.scanWithDuration(progressHandler: { discoveries in
            
        }, completionHandler: { discoveries, error in
            print("\(discoveries.debugDescription), \(error.debugDescription)")
        })
    }
}

extension CenteralController: BKAvailabilityObserver {
    func availabilityObserver(_ availabilityObservable: BKAvailabilityObservable, availabilityDidChange availability: BKAvailability) {
        if availability == .available {
            scan()
        }
    }
}
```

zinwalin, 2021-03-25:
same issue on iPhone XS Max