rhummelmose/BluetoothKit #16: Multiple characteristics per service

cpatterson, 2016-02-10:
If I'm reading things correctly, it looks like BluetoothKit as written only supports a single characteristic per service. Is that correct?

If so, that's a rather crippling limitation on an otherwise well-designed library. I'm working on a project designing a new BLE device and app, with a complex custom BLE service with multiple read/write characteristics.

davidmaitland, 2016-03-12:
I'm wondering the same thing, any response to this yet?

rasmusth, 2016-03-14:
Hi guys.

Sorry for not responding to you any quicker.

Thank you for your interest in the framework and I am glad you like its architecture.

The reason for the fact that the framework only supports its own built-in characteristic, is that it is built for iOS device <-> iOS device communication in a much easier way than achievable using CoreBluetooth. If the framework was to support other custom characteristics, the framework would need added complexity that is not desired for the simpler use case that it was meant for.

closed 2016-03-17 by rasmusth

hisavali, 2017-05-18:
Thanks @rhummelmose for your hard work to create this framework.

It'd would nice to mention above limitations in readme file.

rhummelmose, 2017-05-19:
@hisavali It is stated as the feature of the framework in the readme???

aelam, 2019-02-14:
is there any replacement for this?