rhummelmose/BluetoothKit #18: peripheral.sendData very slow if NSData is larger than a few bytes

matthewkrueger, 2016-03-22:
Hi there - love the library so far.  

Using the example application, I have been experimenting with reading a file to NSData, then sending the NSData using the "sendData" method.

However, if the NSData I'm sending is even a few KB, the transmission takes a very, very long time.  For a 250KB NSData, I'm seeing times around 1-1.5 minutes to do the transfer.  Am I doing something wrong, or is this a limitation of the library?  Thanks!

rasmusth, 2016-03-22:
Unfortunately you'll find that this is a limitation of Bluetooth LE, spiced up by further limitations applied by iOS to the throughput available to each application :)

matthewkrueger, 2016-03-22:
That's unfortunate.  Is a transfer time of 1-1.5 minutes for 250KB of data in line with what you've seen in the past?  I tried slowing increasing the size of my file and it seems the slowdown is linear, so I don't think I'm doing anything especially wrong.  Thanks for the fast reply, much appreciated!

rasmusth, 2016-03-22:
Unfortunately, that sounds like very reasonable throughput. In previous iOS versions this was way worse. My suggestion would be to place the resource on some external location using a network connection and passing an address for it using the Bluetooth link. Of course I don't know your use-case so I'm just guessing away.

matthewkrueger, 2016-03-22:
Unfortunately, the use-case requires transferring a binary file from a device with BLE but no internet connection to an iOS device, and THEN uploading the binary file to the web using the iOS device's internet connection.  Doing a proof of concept using an iPad as the peripheral and iPhone as the central, so I have to pretend like the iPad has no network connection.  Sounds like I'll have to temper the expectations for the transfer speed.

Thanks a lot for the insight and the great library, and especially the functional example project, much appreciated!

rasmusth, 2016-03-22:
You're welcome. Feel free to ask again.

closed 2016-05-12 by rasmusth