rhummelmose/BluetoothKit #26: Two potential bugs.

Nick-The-Uncharted, 2016-06-13:
Thank you for writing this repo which save me a lot of time.  
Here are two potential bug I came across:  
First, `Core Bluetooth` seems to have a slight chance to call `func centralManager(central: CBCentralManager, didConnectPeripheral peripheral: CBPeripheral)` twice for a single connecting behavior.
Code following may output `core bluetooth bug?` when keeping switching on and off bluetooth.

```Swift
internal func centralManager(central: CBCentralManager, didConnectPeripheral peripheral: CBPeripheral) {
        if self.connectionAttempts.count == 0 {
            print("core bluetooth bug?")
        }
        succeedConnectionAttempt(connectionAttemptForPeripheral(peripheral)!)
    }
```

The second one is that if an timeOut error occur . `failConnectionAttempt` will be called, which call `cancelPeripheralConnection` and `cancelPeripheralConnection` lead to call of `didFailToConnectPeripheral `, and when `didFailToConnectPeripheral` happen, the corresponding `connectionAttempt` is already removed and the program will crash .

IMO, both of those two bug can be avoided by avoiding force unwrap result of `connectionAttemptForPeripheral`.

PS: There is a good chance that I mistake something .

rhummelmose, 2016-06-14:
Hi Nick. Thanks for the report, I will try to reproduce and fix it.

rhummelmose, 2016-06-20:
It will probably take me a month before I get time to look at this. Please provide a PR if you're able to fix the issue.

Nick-The-Uncharted, 2016-06-20:
Okay, I will try to find time to do that, I was just not sure whether I mistake something .

commit 28ed220e (2016-08-17): Merge pull request #31 from Nick-The-Uncharted/master

Fix bug cause by force unwrapping.  Fix #26, Fix #28
commit 476d03bd (2016-08-17): Fix bug cause by force unwrapping.  Fix #26
closed 2016-10-21 by rhummelmose