rhummelmose/BluetoothKit #27: If appoint service uuid, it could't find real devices.

pangpingfei, 2016-06-16:
Use "centralManager.scanForPeripheralsWithServices(nil, options: nil)", can discovery all device, but use "centralManager.scanForPeripheralsWithServices(configuration.serviceUUIDs, options: nil)", nothing.

This should be the bug of the official.

rhummelmose, 2016-06-20:
It will probably take me a month before I get time to look at this. Please provide a PR if you're able to fix the issue.

Nick-The-Uncharted, 2016-06-21:
Just make sure your peripheral do boardcasting corresponding service, for example:
```
let dataServiceUUID = NSUUID(UUIDString: Config.sharedInstance.serviceUUID)!
let characteristicUUID = NSUUID(UUIDString: Config.sharedInstance.infoCharacteristicUUID)!
let config = BKPeripheralConfiguration(dataServiceUUID: dataServiceUUID, dataServiceCharacteristicUUID: characteristicUUID)

try peripheral.startWithConfiguration(config)
```
If you are not using this lib for advertising , make sure you have this serviceId in `CBAdvertisementDataServiceUUIDsKey`.

pangpingfei, 2016-06-21:
@Nick-The-Uncharted Thanks, but it seems to have no relationship...

Nick-The-Uncharted, 2016-06-21:
@pangpingfei Actually I manage to discover my device using this lib in way described above.

rhummelmose, 2016-06-21:
@pangpingfei which service UUID are you assigning?

pangpingfei, 2016-06-26:
@rhummelmose "05216AFE-02B0-B11E-3B86-60002A5D5C51"

iCodist, 2017-01-09:
Why can not BKCentral find nearby Bluetooth devices?

rhummelmose, 2017-01-09:
Probably because they are using a different identifier.

iCodist, 2017-01-09:
@rhummelmose How to solve this problem？

LazyyNguyen, 2023-10-04:
@pangpingfei how you solve this prolem?