rhummelmose/BluetoothKit #29: waitting for swift3 support

utopianism, 2016-06-20:
is it have plan for swift3?

rhummelmose, 2016-06-20:
Yep. You can create a PR if you need it within a few months :)

calebkleveter, 2016-09-27:
@utopianism , the version for Swift 3 has been released, but I think you have to get that version through GitHub as it has not been pushed to CocoaPods as far as I know.

    pod 'BluetoothKit', :git => 'https://github.com/rhummelmose/BluetoothKit.git'

rhummelmose, 2016-09-28:
Correct, I will push it to CocoaPods

lou-lan, 2016-10-03:
I still have some mistakes in swift3.
[image]

HoonAhn, 2016-10-07:
When i typed
$ pod install
this error comes up
Analyzing dependencies
[!] The dependency `BluetoothKit (from `https://github.com/rhummelmose/BluetoothKit.git`)` is not used in any concrete target.

what should i do?

calebkleveter, 2016-10-07:
@HoonAhn , you should open a different issue for this.

closed 2017-02-03 by rhummelmose