rhummelmose/BluetoothKit #33: Question: how to find out when a BKRemotePeripheral is ready for send/receive

Daij-Djan, 2016-06-29:
Hi,
I wonder about how to find out when a BKRemotePeripheral is ready for send/receive.

I scanned and discovered it, connected but AFAICS I need to wait for the discovery of its services and characteristics before I sendData..

I wonder if there already is a mechanism or if I should provide a delegate function on BKRemotePeripheralDelegate for it?

Daij-Djan, 2016-06-29:
if I use sendData to early, dataCharateristic is nil - obviously ;) and It fails. btw I made the SendDataTask consider that - before it failed silently.

rhummelmose, 2016-06-29:
Makes sense. Another event should be omitted or it shouldn't make the connected callback before services and characteristics are discovered.

calebkleveter, 2016-09-12:
Seems this thread can be closed now.

closed 2016-09-13 by rhummelmose