rhummelmose/BluetoothKit #42: Question: setup mac as bluetooth keyboard

dicristomanuel, 2016-08-27:
Is it possible with BluetoothKit to vend a bluetooth service from desktop to mobile to use the mac as a keyboard?

rhummelmose, 2016-08-30:
Interesting question. If you need to vend an already defined standard keyboard service, then the answer is no. BluetoothKit defines its own service that is used for transferring arbitrary data.

dicristomanuel, 2016-08-30:
Gotcha. I'm still in the bluetooth learning process but I'm pretty sure that there is a defined keyboard service for the Mac keyboard. Maybe if I find its specifications I can create a service with BluetoothKit? Thanks for your answer @rhummelmose!

rhummelmose, 2016-08-30:
No problem. BluetoothKit is an abstraction that makes it easy to transfer data back and forth between devices, removing the need for dealing with services, characteristics, descriptors and so forth. Unfortunately this also means that you won't be able to use BluetoothKit for what you're aiming for.

closed 2016-08-30 by dicristomanuel