rhummelmose/BluetoothKit #44: Scanning false when config with CBUUID  example: "0xCBBB"

buidinhduong, 2016-09-07:
instead of configuration with NSUUID 128 bit (example : "**6E6B5C64-FAF7-40AE-9C21-**D4933AF45B23"), i want to set up with CBUUID like  "**0xCBBB**" but the Central scanning interrupted  many times. and can not sending data.
Please help

rhummelmose, 2016-09-08:
Hi.

Why do you want to set it up with a CBUUID?

If necessary you can convert it from a CBUUID to and NSUUID. See the following for reference:
http://stackoverflow.com/questions/36212020/how-can-i-convert-a-bluetooth-16-bit-service-uuid-into-a-128-bit-uuid

dorucazan, 2017-07-21:
Hi,
I had same issue since I am trying to work with a HC-05 bluetooth module. I kind of "solve" it overloading init in BKConfiguration class with CBUUID params (-: 

```
    public init(dataServiceUUID: CBUUID, dataServiceCharacteristicUUID: CBUUID) {
        self.dataServiceUUID = dataServiceUUID
        self.dataServiceCharacteristicUUID = dataServiceCharacteristicUUID
        endOfDataMark = "EOD".data(using: String.Encoding.utf8)!
        dataCancelledMark = "COD".data(using: String.Encoding.utf8)!
    }
```
and then use it like:

```
    let dataServiceUUID = CBUUID(string: "FFE0")
    let dataServiceCharacteristicUUID = CBUUID(string: "FFE1")
    let configuration = BKConfiguration(dataServiceUUID: dataServiceUUID, dataServiceCharacteristicUUID: dataServiceCharacteristicUUID)
```

Steve-001, 2020-06-03:
mark