rhummelmose/BluetoothKit #45: Question: Is it possible to print barcode/qr code using Bluetooth Kit?

yhelfronda, 2016-09-08:
Hi,

Is it possible to print qr code/barcode using this library? If so, can you please guide me 'cause we are planning to make an app which has feature like qr code/barcode printing. Or it's a pleasure if you can provide an example?

Many thanks!

rhummelmose, 2016-09-08:
Are you kidding me?

yhelfronda, 2016-09-08:
Nope.

rhummelmose, 2016-09-08:
Have you read the readme?

yhelfronda, 2016-09-08:
Yeah! But how? The given example is just texts.

rhummelmose, 2016-09-08:
BluetoothKit is an abstraction over CoreBluetooth. Why do you think it supports anything related to QR-codes and barcodes?

calebkleveter, 2016-09-08:
Is the question "Can you send QR and bar codes to other devices using this library?"?

yhelfronda, 2016-09-09:
@calebkleveter What I mean is, can you print qr or bar code to bluetooth printing device instead of just printing 'texts', just like printing receipts.

rhummelmose, 2016-09-09:
Sorry but you are misunderstanding what this framework is trying to solve. What you need is CoreBluetooth and a description of the service on the printing device that you are trying to communicate with.

lou-lan, 2016-10-03:
You need to make an agreement with your hardware so that you can quickly develop。