rhummelmose/BluetoothKit #53: Swift3 branch requires iOS 10+ now

3ph, 2016-10-06:


calebkleveter, 2016-10-06:
Yes. Apple merged the two classes `CBPeripheralManager` and `CBCentralManager` in to one class, `CBManager`. `CBManager` is only available in iOS 10, so you are going to need that latest iOS SDK to develop with Bluetooth. This means you will need to install Xcode 8.0.

3ph, 2016-10-06:
Hi, can you point me to where did you find this? Only changes in API I could find is that CBManager is new superclass of the CBPeripheralManager and CBCentralManager. The state was moved there, nothing else.

Neither of these classes were deprecated. Could find any reason why the drastic change of deployment target was necessary?

Thanks

mrdavidrees, 2016-10-18:
resolved in #54

rhummelmose, 2016-10-21:
Thanks @mrdavidrees!

closed 2016-10-21 by rhummelmose