rhummelmose/BluetoothKit #57: Couldn't work properly on iOS 10.

SelfnessAid, 2016-11-07:
I'd like to use this framework for my app, but I faced following issue on IOS10 with Swift3.0.

> **Error from scanning: The operation couldn't be completed. (BluetoothKit.BKError error 3.)**

Also I checked this on iOS 9(XCode 7.3, iPhone 5s, Gimbal Beacon Series 21) , but occurred the same issue.
(I tried with this framework [https://github.com/tinydream12/Bluetoothkit](https://github.com/tinydream12/Bluetoothkit), but the same result.)

Does anyone know how to avoid this?
Thank you.

rhummelmose, 2016-11-08:
Please show your code?

SelfnessAid, 2016-11-10:
- _CentralViewController.swift_ in the BKExample
```
    func scan() {
        central.scanContinuouslyWithChangeHandler({ changes, discoveries in
            let indexPathsToRemove = changes.filter({ $0 == .remove(discovery: nil) }).map({ NSIndexPath(row: self.discoveries.index(of: $0.discovery)!, section: 0) })
            self.discoveries = discoveries
            let indexPathsToInsert = changes.filter({ $0 == .insert(discovery: nil) }).map({ NSIndexPath(row: self.discoveries.index(of: $0.discovery)!, section: 0) })
            if !indexPathsToRemove.isEmpty {
                self.discoveriesTableView.deleteRows(at: indexPathsToRemove as [IndexPath], with: UITableViewRowAnimation.automatic)
            }
            if !indexPathsToInsert.isEmpty {
                self.discoveriesTableView.insertRows(at: indexPathsToInsert as [IndexPath], with: UITableViewRowAnimation.automatic)
            }
            for insertedDiscovery in changes.filter({ $0 == .insert(discovery: nil) }) {
                Logger.log("Discovery: \(insertedDiscovery)")
            }
        }, stateHandler: { newState in
            if newState == .scanning {
                self.activityIndicator?.startAnimating()
                return
            } else if newState == .stopped {
                self.discoveries.removeAll()
                self.discoveriesTableView.reloadData()
            }
            self.activityIndicator?.stopAnimating()
        }, errorHandler: { error in
            Logger.log("Error from scanning: \(error.localizedDescription)")
        })
    }
```

- _BKCentral.swift_ in the BluetoothKit

BottleMan, 2016-11-28:
I have just met the same error, and when I dig into the code and find out that the state is 'scanning' in this function:

```swift
    private func handleScanEvent(_ event: Event) throws {
        switch state {
        case .available:
            state = .scanning
        default:
            throw BKError.transitioning(currentState: state, validStates: [ .available ])
        }
    }
```

in BluetoothKit > Source > BKCentralStateMachine.swift > line99.

Could anyone give some advice?

rhummelmose, 2016-11-28:
Isn't it just calling scan twice? I think it is just the example app that does that, it is no problem.

rhummelmose, 2016-11-30:
@BottleMan ?

BottleMan, 2016-12-02:
@rhummelmose sorry for replying late, I just comment the 'scan' method in viewDidAppear and it goes alright.

```swift
    internal override func viewDidAppear(_ animated: Bool) {
        //scan()
    }
```

rhummelmose, 2016-12-02:
Then it won't always start scanning when you navigate back and forth. That is why it is there :)