rhummelmose/BluetoothKit #58: fail to build with carthage

olivier38070, 2016-11-13:
Hi,

using this line, in my cart file  
github "rasmusth/BluetoothKit" ~> 0.2.0
using carthage v 0.18.1, Xcode 8.1, Sierra 10.12.1

fail, with a dependency error code 65.
the log is long, I can provide it, if needed.

I noticed that in the checkout folder, the bluetooth project seem to be in swift 2...

if I download the project, manually, from github : it compile straight away.

thanks for any help
Olivier

rhummelmose, 2016-11-14:
It is due to the fact that no new version has been released. I will do it within a weeks time. Sorry for the inconvenience.

DerekK19, 2016-11-15:
I had the same, so I referenced the development branch directly:

github "rasmusth/BluetoothKit" "development"