rhummelmose/BluetoothKit #59: Peripheral switches off, Central doesn't react correctly

DerekK19, 2016-11-15:
My Peripheral is a Mac app and my Central is an iOS app. I would think that when my central is connected to a peripheral, if that peripheral is switched off (i.e I quite the Mac app) the central should recognise the peripheral has disconnected.

In fact the following is logged in my Xcode console when I'm debugging my Central

2016-11-15 19:01:39.337791 MyApp[5707:2457596] [CoreBluetooth] WARNING: The delegate for <CBPeripheral: 0x1742e2d00, identifier = <uuid>, name = <name, state = connected> does not implement -[peripheral:didModifyServices:]

I have removed my UIUID and my name from the message

I'm using MacOS Sierra, Xcode 8.1 and Swift 3

rhummelmose, 2016-11-15:
@DerekK19 I will look into this now. I haven't tested on macOS, seems it behaves differently from iOS and tvOS. Sorry.

rhummelmose, 2016-11-15:
@DerekK19 realising now that I'll need an example project for Mac. Will probably take a day or two.