rhummelmose/BluetoothKit #62: InternalError

tomneo2004, 2016-12-15:
I try out BluetoothKit. I encounter that I always get InternalError(BKError) in callback after trying to scan Peripheral. I have no idea where it goes wrong? Can anyone help?  Below code is in UIViewController

`

    let peripheral : BKPeripheral = BKPeripheral()
    let central : BKCentral = BKCentral()
    
    override func viewDidLoad() {
        super.viewDidLoad()
        // Do any additional setup after loading the view, typically from a nib.
    }

    override func didReceiveMemoryWarning() {
        super.didReceiveMemoryWarning()
        // Dispose of any resources that can be recreated.
    }

    @IBAction func server(){
        
        self.peripheral.delegate = self
        
        do {
            let serviceUUID = NSUUID(uuidString: "6E6B5C64-FAF7-40AE-9C21-D4933AF45B23")!
            let characteristicUUID = NSUUID(uuidString: "477A2967-1FAB-4DC5-920A-DEE5DE685A3D")!
            let localName = "My Cool Peripheral"
            let configuration = BKPeripheralConfiguration(dataServiceUUID: serviceUUID as UUID, dataServiceCharacteristicUUID:  characteristicUUID as UUID, localName: localName)
            try self.peripheral.startWithConfiguration(configuration)
            // You are now ready for incoming connections
        } catch let error {
            // Handle error.
            
            NSLog(error as! String)
        }
    }
    
    @IBAction func client(){
        
        
        self.central.delegate = self
        central.addAvailabilityObserver(self)
        do {
            let serviceUUID = NSUUID(uuidString: "6E6B5C64-FAF7-40AE-9C21-D4933AF45B23")!
            let characteristicUUID = NSUUID(uuidString: "477A2967-1FAB-4DC5-920A-DEE5DE685A3D")!
            let configuration = BKConfiguration(dataServiceUUID: serviceUUID as UUID, dataServiceCharacteristicUUID: characteristicUUID as UUID)
            try central.startWithConfiguration(configuration)
            // Once the availability observer has been positively notified, you're ready to discover and connect to peripherals.
        } catch let error {
            // Handle error.
            NSLog(error as! String)
        }
    }

    func peripheral(_ peripheral: BKPeripheral, remoteCentralDidConnect remoteCentral: BKRemoteCentral) {
        
    }
    
    func peripheral(_ peripheral: BKPeripheral, remoteCentralDidDisconnect remoteCentral: BKRemoteCentral) {
        
    }
    
    func central(_ central: BKCentral, remotePeripheralDidDisconnect remotePeripheral: BKRemotePeripheral) {
        
    }
    
    func availabilityObserver(_ availabilityObservable: BKAvailabilityObservable, availabilityDidChange availability: BKAvailability) {
        
        if availability == .available{
            self.central.scanContinuouslyWithChangeHandler({ changes, discovers in
                
                NSLog("\(discovers)")
                }, stateHandler: { newState in
                    
                }, errorHandler: { error in
                    
                    NSLog("error \(error)")
            })
        }else{
            
            self.central.interruptScan()
        }
    }

`

donly, 2017-03-29:
same issue here

rhummelmose, 2017-03-29:
@tomneo2004 which error do you get?