rhummelmose/BluetoothKit #67: Cleaner access to Characteristics & Changes

foreignmedia, 2017-03-08:
Would be great if we could access the characteristics and get change notifications via the BKRemotePeerDelegate. Right now it looks like the configuration is locked down. This would help greatly with more advanced devices that switch characteristics based on the mode they're in.

rhummelmose, 2017-03-08:
The device you talk about, is that another Apple device running BluetoothKit? Otherwise you are probably looking at the wrong framework. It would require extensive work to make the framework useful for arbitrary devices and then there wouldn't be much simplicity gain over Apple's CoreBluetooth.

foreignmedia, 2017-03-08:
Apple and others. Not looking for anything crazy, just access to the advertised characteristics array of a given remote. Great lib, by the way.

rhummelmose, 2017-03-13:
Thank you. You are correct in that the lib could be modified to support this, the only thing you need to be aware of is that the lib also has a level of logic in place to chunk data if it is larger than what is possible in current context. If you have time available you are more than welcome to help implementing and creating a pull request.