rhummelmose/BluetoothKit #70: Delay of Sending or Receiving data to BLE Device?

nladuo, 2017-03-26:
I am using the BLE CC2541 module to interact with my arduino device. 

For testing, I connected the TxD pin and RxD pin directly for that I can get the same data when I send data in iOS App.

Like this:
[image]

First, I used the Example of this repository. After I modified the UUID of service and characteristic, I successfully scanned my BLE device. But when I send multiple data, it only received one.
[image]

And then I created a simple project, It has a button for sending data, a textfield for what to send and a textview for displaying the received data. 

The code is like this:
``` swift
    @IBAction func onSendBtnClicked(_ sender: Any) {
        
        let data: Data = (inputTextField.text?.data(using: String.Encoding.utf8))!
        
        
        central.sendData(data, toRemotePeer: remotePeripheral) { data, remotePeripheral, error in
            guard error == nil else {
                print("Failed sending to \(remotePeripheral)")
                return
            }
            print("Sent to \(remotePeripheral)")
        }
    }

    func remotePeer(_ remotePeer: BKRemotePeer, didSendArbitraryData data: Data) {
        print("Recieved: \(String(data: data, encoding: String.Encoding.utf8) ?? "error")")
        
        receivedTextView.text = receivedTextView.text + String(data: data, encoding: String.Encoding.utf8)!
    }
```
I filled the TextField with "gghh", after multiple clicking on sendBtn, what it returned like this:
[image]

Previously, I used a objective-c library: [SerialGATT.h](https://github.com/nladuo/ElectronicDIY/blob/master/03-PM2.5%E6%A3%80%E6%B5%8B%E4%BB%AA/BlePM2.5_iOS/BleCom/SerialGATT.h), [SerialGATT.m](https://github.com/nladuo/ElectronicDIY/blob/master/03-PM2.5%E6%A3%80%E6%B5%8B%E4%BB%AA/BlePM2.5_iOS/BleCom/SerialGATT.m)
 which can receive the data at real time when I tap the button.
I am using `pod 'BluetoothKit', '~> 0.4.0'`.

rhummelmose, 2017-03-29:
Who is sending the data that you receive there, the external device? I would suggest that you look into why it only sends you that data once.

nladuo, 2017-03-30:
Yes, it is a external device, a [UART Transceiver Serial Module](https://www.amazon.com/JT-Tech-Bluetooth-Transceiver-Baseboard/dp/B01MZBS4FD/ref=sr_1_4?ie=UTF8&qid=1490840541&sr=8-4&keywords=uart+ble). When I connected the RxD Pin and the TxD Pin and then sent data to this device,  this device would send the data what it had received simultaneously.

rhummelmose, 2017-03-30:
Could be that the device doesn't support receiving as big chunks as the iOS device.

nladuo, 2017-03-31:
It has a objective-c based sample and works well. I want to migrate it to swift-based one.

``` obj-c
-(void) writeValue:(int)serviceUUID characteristicUUID:(int)characteristicUUID p:(CBPeripheral *)p data:(NSData *)data {
    UInt16 s = [self swap:serviceUUID];
    UInt16 c = [self swap:characteristicUUID];
    NSData *sd = [[NSData alloc] initWithBytes:(char *)&s length:2];
    NSData *cd = [[NSData alloc] initWithBytes:(char *)&c length:2];
    CBUUID *su = [CBUUID UUIDWithData:sd];
    CBUUID *cu = [CBUUID UUIDWithData:cd];
    CBService *service = [self findServiceFromUUIDEx:su p:p];
    if (!service) {
        printf("Could not find service with UUID %s on peripheral with UUID %s\r\n",[self CBUUIDToString:su],[self UUIDToString:(__bridge CFUUIDRef )p.identifier]);
        return;
    }
    CBCharacteristic *characteristic = [self findCharacteristicFromUUIDEx:cu service:service];
    if (!characteristic) {
        printf("Could not find characteristic with UUID %s on service with UUID %s on peripheral with UUID %s\r\n",[self CBUUIDToString:cu],[self CBUUIDToString:su],[self UUIDToString:(__bridge CFUUIDRef )p.identifier]);
        return;
    }
    if(characteristic.properties & CBCharacteristicPropertyWriteWithoutResponse)
    {
        [p writeValue:data forCharacteristic:characteristic type:CBCharacteristicWriteWithoutResponse];
    }else
    {
        [p writeValue:data forCharacteristic:characteristic type:CBCharacteristicWriteWithResponse];
    }
}
```

nhatlee, 2017-04-05:
I faced the same problem when try to send photos between two devices.It's 50% failed, and need to wait so long to receive the photo.

rhummelmose, 2017-04-05:
@nhatlee BLE isn't made for transferring large files, hence the speed. I haven't encountered these issues, I will try to reproduce.