rhummelmose/BluetoothKit #103: Example not working

MHX792, 2020-09-11:
See screenshot:

[image]

beilusen, 2020-09-21:
You need to add the framework prompted by the error in General

bluce-clj, 2020-11-10:
[16:13:49.473] Error from scanning: internalError(underlyingError: Optional(BluetoothKit.BKCentralStateMachine.BKError.transitioning(currentState: BluetoothKit.BKCentralStateMachine.State.scanning, validStates: [BluetoothKit.BKCentralStateMachine.State.available])))

bluce-clj, 2020-11-10:
add third framework fix this prompted, but run center mode show error.

zinwalin, 2021-03-25:
[16:52:21.732] Error from scanning: internalError(underlyingError: Optional(BluetoothKit.BKCentralStateMachine.BKError.transitioning(currentState: BluetoothKit.BKCentralStateMachine.State.scanning, validStates: [BluetoothKit.BKCentralStateMachine.State.available])))[16:52:24.483]