rhummelmose/BluetoothKit #108: pod install

AMBIENTE1, 2021-01-12:
my Podfile:
source 'https://github.com/CocoaPods/Specs.git'
platform :ios, '10.0'
use_frameworks!

pod 'BluetoothKit', '~> 0.4.0'

Installing BluetoothKit (0.4.0)
[!] Unable to determine Swift version for the following pods:

- `BluetoothKit` does not specify a Swift version and none of the targets (`Pods`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod.

[!] The abstract target Pods is not inherited by a concrete target, so the following dependencies won't make it into any targets in your project:
    - BluetoothKit (~> 0.4.0)

JeremiSharkboy, 2021-06-21:
ENV['SWIFT_VERSION'] = '5'
source 'https://github.com/CocoaPods/Specs.git'
platform :osx, '10.10'
target 'Nine.CLP' do
use_frameworks!

pod 'BluetoothKit'

end
    post_install do |installer|
        installer.pods_project.targets.each do |target|
            target.build_configurations.each do |config|
                config.build_settings['SWIFT_VERSION'] = '5.0'
            end
        end
    end

rhummelmose, 2021-07-28:
This should work with newest version