rhummelmose/BluetoothKit #110: Error nil in central.scanWithDuration

ironrustem, 2021-03-26:
I have problem. I scan with central.scanWithDuration, but I get nil error. Can you help me please? Why?

My code:`

    private var discoveries = [BKDiscovery]()
    let central = BKCentral()

    func central(_ central: BKCentral, remotePeripheralDidDisconnect remotePeripheral: BKRemotePeripheral) {
        print("DidDisconnect")
    }

    func availabilityObserver(_ availabilityObservable: BKAvailabilityObservable, availabilityDidChange availability: BKAvailability) {
        if availability == .available {
            print("start scan")
            scan()
        } else {
            central.interruptScan()
        }
    }

    func availabilityObserver(_ availabilityObservable: BKAvailabilityObservable, unavailabilityCauseDidChange unavailabilityCause: BKUnavailabilityCause) {
        print("CauseDidChange")
    }

    override func viewDidLoad() {
        super.viewDidLoad()
        
        tableView.delegate = self
        tableView.dataSource = self

        startCentral()
    }

    private func startCentral() {
        do {
            central.delegate = self
            central.addAvailabilityObserver(self)
            let dataServiceUUID = UUID(uuidString: "6E6B5C64-FAF7-40AE-9C21-D4933AF45B23")!
            let dataServiceCharacteristicUUID = UUID(uuidString: "477A2967-1FAB-4DC5-920A-DEE5DE685A3D")!
            let configuration = BKConfiguration(dataServiceUUID: dataServiceUUID, dataServiceCharacteristicUUID: dataServiceCharacteristicUUID)
            try central.startWithConfiguration(configuration)
        } catch let error {
            print("Error while starting: \(error)")
        }
    }

    private func scan() {
        central.scanWithDuration(7, progressHandler: { newDiscoveries in
            print(newDiscoveries)
        }, completionHandler: { result, error in
           print("errorScan \(error)")
            print(result)
        })
    }`

rhummelmose, 2021-07-28:
Check the example please :)

643824979, 2021-12-23:
The example also returns an error