rhummelmose/BluetoothKit #115: Crashes when send large data.

tianzhuqiao, 2022-01-03:
Thanks for the great project. It works perfectly when sending small amount of data (e.g., < 1k). However, crashes when sending large data (e.g., ~1M bytes). Looks like the `processSendDataTasks` function used in `sendData` is recursive (e.g., on my phone, each call sends 182 bytes). Does it mean that it doesn't support sending large data (e.g., ~M bytes) (limited by the stack size)? 

Why do we use a recursive function to send all data? Will a "for" loop work? Thanks!

https://github.com/rhummelmose/BluetoothKit/blob/2f268e43495db001ae92d9fedf1982e0e69d29da/Source/BKPeer.swift#L67

hrishav503, 2022-06-03:
did you find any solution sir?

tianzhuqiao, 2022-06-03:
Not really. Tried to replace the recursive function with a for loop; but for some unknown reason, it only runs intermittently.